fbiego/ESP32_BLE_OTA_Arduino #2: App crash after file select

ericlangel, 2021-04-08:
Hello,

i try to test your BLE OTA example.

But the App Crash(close) after selecting the File. I Also get a info that the App is crashed.

Android 10
Xiaomi Note 10 Lite

ashish-iottive, 2021-04-08:
Same here Application Crashes while uploading BIN File

fbiego, 2021-04-08:
The issue is caused by changes in Storage Access Framework starting from Android 10. I'm working to resolve the issue as soon as possible

ericlangel, 2021-04-08:
yeah, i suspected this

in my own app i made a folder on external Storage, for Android 10 and 11 its working.
So that the .bin cant inserted everywere....just in this folder.
and with a dropdown you could list the available files in that folder

fbiego, 2021-04-09:
Tested on an Emulator. Let me know if it works

ericlangel, 2021-04-09:
short test on Andorid 10 (without ble) was ok

i'll check it later with android 11 and also BLE

ericlangel, 2021-04-14:
So, i got some time to test the example

the BLE Data Transfer seems to be working on Android 10.

But....nothing updates on the ESP32

Command 0xFD, 0xFF, and 0xFB are transmitted
0xFB with a length of 100 -> why not 512?

and after many 0xFB's it just disconnect

so update.bin is never written
and esp32 is never rebooted

ericlangel, 2021-05-21:
can be closed

closed 2021-05-21 by ericlangel