fbiego/ESP32_BLE_OTA_Arduino #4: How to BLE OTA Functionality in this code ?

IoTopsDeveloper, 2021-05-12:
Hey @fbiego I am IoT Developement here your recent commenter on your recent Youtube video. Below here is my code in which I want to add BLE OTA part, and I am confused how to add it with the help of your code.Kindly look into it.
` // Default Arduino includes
#include <Arduino.h>
#include <WiFi.h>
#include <WiFiMulti.h>
#include <nvs.h>
#include <nvs_flash.h>
#include <ArduinoJson.h>
#include <dht.h>
#define dht_dpin 18  //no ; here. Set equal to channel sensor is on
dht DHT;
#include <BLEUtils.h>
#include <BLEServer.h>
#include <BLEDevice.h>
#include <BLEAdvertising.h>
#include <BLE2902.h>
#include <Preferences.h>
int LEDBuiltInConnected=4;
int LEDBLEMode=22;
int LEDWifiNotConnected=23;
int Button1=32;
int Button2=35;
int LED1 = 19;
int LED2 = 21;
char previous_state[4];
char globalJson[200]; 
char wifi_status[30];
char Nill[150];
WiFiMulti wifiMulti;
char apName[] = "ESP32 AUTO";
const char compileDate[] = __DATE__ " " __TIME__;

bool hasCredentials = false;
volatile bool isConnected = false;
bool connStatusChanged = false;

// List of Service and Characteristic UUIDs
#define SERVICE_UUID  "0000aaaa-ead2-11e7-80c1-9a214cf093ae"
#define WIFI_UUID     "00005555-ead2-11e7-80c1-9a214cf093ae"
#define CONTROL_UUID  "6b464d27-c2da-4df7-8eb9-81357ad247d5"
#define NOTIFY_UUID   "1404fd3e-979e-43de-bdb2-6234c9ec8cbe"

/** SSIDs of local WiFi networks */
String ssidPrim;
String ssidSec;
/** Password for local WiFi network */
String pwPrim;
String pwSec;

/** Characteristic for digital output */
BLECharacteristic *pCharacteristicWiFi;
BLECharacteristic *pCharacteristic3;
/** BLE Advertiser */
BLEAdvertising* pAdvertising;
/** BLE Service */
BLEService *pService;
/** BLE Server */
BLEServer *pServer;

/** Buffer for JSON string */
// MAx size is 51 bytes for frame: 
// {"ssidPrim":"","pwPrim":"","ssidSec":"","pwSec":""}
// + 4 x 32 bytes for 2 SSID's and 2 passwords
StaticJsonBuffer<200> jsonBuffer;
StaticJsonBuffer<200> JSONBuffer;
/**
 * MyServerCallbacks
 * Callbacks for client connection and disconnection
 */

[... 339 lines cut ...]
String chunk1=Gjson.substring(0,20);
String chunk2=Gjson.substring(21,40);
String chunk3=Gjson.substring(41,60);
String chunk4=Gjson.substring(61,80);
String chunk5=Gjson.substring(81,100);
String chunk6=Gjson.substring(101,120);
chunk1.getBytes(c1,20);
chunk2.getBytes(c2,20);
chunk3.getBytes(c3,20);
chunk4.getBytes(c4,20);
chunk5.getBytes(c5,20);
chunk5.getBytes(c6,20);
pCharacteristic3->setValue((char*) c1);
pCharacteristic3-> notify ();
pCharacteristic3->setValue((char*) c2);
pCharacteristic3-> notify ();
pCharacteristic3->setValue((char*) c3);
pCharacteristic3-> notify ();
pCharacteristic3->setValue((char*) c4);
pCharacteristic3-> notify ();
pCharacteristic3->setValue((char*) c5);
pCharacteristic3-> notify ();
pCharacteristic3->setValue((char*) c6);
pCharacteristic3-> notify ();
}`

fbiego, 2021-05-13:
Do you have a GitHub repository? That would be much easier

IoTopsDeveloper, 2021-05-13:
> Do you have a GitHub repository? That would be much easier

https://github.com/IoTopsDeveloper/BLEwithWifi

IoTopsDeveloper, 2021-05-13:
@fbiego any guidance or anything adding in  the Arduino Sketch ?

IoTopsDeveloper, 2021-05-13:
@fbiego any guidance or anything adding in  the Arduino Sketch ?
 
Or the OTA part in your firmware requires Service UUID with Characterstics UUID for Update and Installation or else Service UUID for Service and both CHARACTERSTIC_UUID_RX and CHARACTERISTIC_UUID_TX are the one which perform operation ?

fbiego, 2021-05-13:
I'm checking it now

fbiego, 2021-05-13:
Check the PR on your [`repo`](https://github.com/IoTopsDeveloper/BLEwithWifi/pull/1)

IoTopsDeveloper, 2021-05-27:
Actually I find the problem in the firmware when I implemented my code with BLE OTA where `writeBinary` function and  `updateFromFS` has been run into `void loop` function but the OTA part always run in the `void setup`. Is that right ?

fbiego, 2021-05-27:
Have you tried out the bare BLE OTA sketch without adding your code?
Does it work?

IoTopsDeveloper, 2021-05-27:
It worked but in Serial Monitor and in Application their is error 9 displaying.

Actually my point was that the BLE OTA withmy code crashes get occured due to both Wifi part and `Update`, `FS` , `FFat` and `SPIFF`, all run in same `void loop`  And adding Wifi functions in `void loop` itself creates an infinite loop, without working of all 'Update` `FS` `FFat` `SPIFF` of OTA Part, this is where it created problem.

You make very good code and I am really appreciating your efforts you have done with Arduino Sketch and Android Application. Infact your updated application V1.3 worked really well with any BIN file .

It would be very helpful if you provide the Source Code for [DIY ESP 32 Clock with BLE OTA ](https://www.youtube.com/watch?v=TU_O4UPm00A) at your own will. Perhaps the last hope may help me to solve my problem.
Or I will try try and try more.

Apologies if any inconvenience caused

fbiego, 2021-05-27:
The clock sketch is based on BLE OTA, it has additional functionality for time, LEDs, etc.
The file system used is FFat. 

The BLE OTA sketch should be a good start then add your desired functionality

```
#include <Update.h>
#include "FS.h"
#include "FFat.h"
#include <BLEDevice.h>
#include <BLEUtils.h>
#include <BLEServer.h>
#include <BLE2902.h>
#include <ESP32Time.h>
#include <WS2812FX.h>
//#include "ESP_RMT_Driver.h"
//#include "start.h"

#define LED_PIN     23
#define LED_COUNT  28
#define BRIGHTNESS 50
#define OFF_BUTTON  32

#define FORMAT_FFAT false
#define FLASH FFat

#define NORMAL_MODE   0
#define UPDATE_MODE   1
#define OTA_MODE      2

uint8_t major = 1;
uint8_t minor = 6;
uint8_t ver[] = {0xFA, major, minor};  // code version

uint8_t updater[16384]; // >= MainActivity.PART
uint8_t updater2[16384];
static uint32_t color = GREEN;
static uint32_t color2 = BLUE;
static uint32_t clr = 0x000000;

uint8_t digits[11] = {
  0x77,       //  0
  0x44,       //  1
  0x3E,       //  2
  0x6E,       //  3
  0x4D,       //  4
  0x6B,       //  5
  0x7B,       //  6
  0x46,       //  7
  0xFF,       //  8
  0x6F,       //  9
  0x00        //  blank
};

#define SERVICE_UUID              "fb1e4001-54ae-4a28-9f74-dfccb248601d"
#define CHARACTERISTIC_UUID_RX    "fb1e4002-54ae-4a28-9f74-dfccb248601d"
#define CHARACTERISTIC_UUID_TX    "fb1e4003-54ae-4a28-9f74-dfccb248601d"

static BLECharacteristic* pCharacteristicTX;
static BLECharacteristic* pCharacteristicRX;

WS2812FX ws2812fx = WS2812FX(LED_COUNT, LED_PIN, NEO_GRB  + NEO_KHZ800);
ESP32Time rtc;

static bool deviceConnected = false;
static int id = 0;
static bool writeFile = false;
static int writeLen = 0;
static int writeLen2 = 0;
static bool current = true;
[... 413 lines cut ...]
      return;
    }

    size_t updateSize = updateBin.size();

    if (updateSize > 0) {
      Serial.println("Trying to start update");
      performUpdate(updateBin, updateSize);
    }
    else {
      Serial.println("Error, file is empty");
    }

    updateBin.close();

    // when finished remove the binary from spiffs to indicate end of the process
    Serial.println("Removing update file");
    fs.remove("/update.bin");

    rebootEspWithReason("Rebooting to complete OTA update");
  }
  else {
    Serial.println("Could not load update.bin from spiffs root");
  }
}```

IoTopsDeveloper, 2021-06-01:
@fbiego  Thank You very much for providing the source code. It definitely help a-lot to understand the problem. But their is one error 9 displaying every time Sketch been uploaded. Is the size of the BIN file is set below 1MB in the sketch or we can upload upto 2-3 MB in the `updater` section

fbiego, 2021-06-01:
Which partition scheme are you using?
error 9 is defined as UPDATE_ERROR_ACTIVATE  ("Could Not Activate The Firmware") not sure what is causing that

IoTopsDeveloper, 2021-06-01:
Customized Partiton as defined in the article [Custom Partition Scheme](https://desire.giesecke.tk/index.php/2018/04/20/change-partition-size-arduino-ide/)

fbiego, 2021-06-01:
Can you define your partitions and sizes?

it is up to you to decide where to run the update from, `void setup` or `void loop`. Loop is convenient because the update can be started right after the file is received otherwise you would have to reboot to update it from setup.
I don't see why you should be running other tasks when updating

IoTopsDeveloper, 2021-06-01:
> it is up to you to decide where to run the update from, `void setup` or `void loop`. Loop is convenient because the update can be started right after the file is received otherwise you would have to reboot to update it from setup.
> I don't see why you should be running other tasks when updating

I had made different Partition Scheme for the my previous code as mention in this article [Custom Partition Scheme](https://desire.giesecke.tk/index.php/2018/04/20/change-partition-size-arduino-ide/) . How could I run update section in `void loop` with BLE ?
 
If I can run two seperate tasks in parallel using FreeRTOS i.e Wifi Running with MQTT and BLE Firmware Update  or  Disconnect Wifi when OTA Update starts ?

IoTopsDeveloper, 2021-06-07:
Which type of Partition Scheme you used while uploading the Firmware sketch via Arduino IDE at which Baud Rate , Flash Size and Core Debug Level  ?

KordelFranceTech, 2021-08-10:
> Which type of Partition Scheme you used while uploading the Firmware sketch via Arduino IDE at which Baud Rate , Flash Size and Core Debug Level ?

Yes, can you indicate which partition scheme, flash size, baud rate, etc. you are using to build your "firmware.bin" file? I am having issues with this as well. Your "firmware.bin" file transfers perfectly fine, but I cannot get any other lightweight .bin file to transfer over.

fbiego, 2021-08-11:
I have tested on two boards using Arduino IDE.
The settings are defaults including upload speed which is 921600. The only changes are as below;

## Board 1
- Board: DOIT ESP32 DEVKIT V1
The partition type will be SPIFFS, so in the code,
`#define USE_SPIFFS  //comment to use FFat`

## Board 2
- Board: ESP32 Dev Module
- Flash Size: 16MB (128Mb)
- Partiton Scheme: 16M Flash(3MB APP/9MB FATFS)
The partition type will be FAT
`//#define USE_SPIFFS  //comment to use FFat`