fbiego/ESP32_BLE_OTA_Arduino #5: ESP32 with additional BLE-characteristics

goosst, 2021-06-07:
Hello

Thanks for this great implementation.
Your test example works great, so I was trying to integrate your code into my mine to have OTA_BLE functionality.
I might be doing something I'm not fully understanding, so apologies in advance:

My goal is to post some values in a few BLE-characteristics. After integrating the BLE-OTA, I seem to fail at this. I might have butchered something in your initBLE() function (or didn't understand it).

It seems like the additional characteristics are just not available at all anymore (when scanning through my phone). 
And every time I called `pAdvertising->start();` in the loop() the ESP32 reboots.

In case you have any suggestions, they are very welcome.

thanks!

```
BLEServer *pServer;
BLEService *pService;
BLEAdvertising *pAdvertising;

// BluetoothSerial ESP_BT; //Object for Bluetooth
BLECharacteristic *pCharacteristic1;
BLECharacteristic *pCharacteristic2;
BLECharacteristic *pCharacteristic3;
BLECharacteristic *pCharacteristic4;

static BLECharacteristic *pCharacteristicTX;
static BLECharacteristic *pCharacteristicRX;

#define SERVICE_UUID "4fafc201-1fb5-459e-8fcc-c5c9c331914b"
#define CHARACTERISTIC_UUID "beb5483e-36e1-4688-b7f5-ea07361b26a8"  //power
#define CHARACTERISTIC2_UUID "688091db-1736-4179-b7ce-e42a724a6a68" //battery voltage
#define CHARACTERISTIC3_UUID "d240dbed-7d22-45bb-b810-add58a6c856b" // rpm
#define CHARACTERISTIC4_UUID "41e5e3f7-47e2-4885-945c-9dda1fc1dc7c" // operating state

// #define SERVICE_UUID2 "fb1e4002-54ae-4a28-9f74-dfccb248601d"
#define CHARACTERISTIC_UUID_RX "fb1e4002-54ae-4a28-9f74-dfccb248601d"
#define CHARACTERISTIC_UUID_TX "fb1e4003-54ae-4a28-9f74-dfccb248601d"
```

```
void initBLE()
{
  BLEDevice::init("ESP32 Hover");
  // BLEServer *pServer = BLEDevice::createServer();
  pServer = BLEDevice::createServer();
  pServer->setCallbacks(new MyServerCallbacks());

  // BLEService *pService2 = pServer->createService(SERVICE_UUID2);
  pService = pServer->createService(SERVICE_UUID);
  pCharacteristicTX = pService->createCharacteristic(CHARACTERISTIC_UUID_TX, BLECharacteristic::PROPERTY_NOTIFY);
  pCharacteristicRX = pService->createCharacteristic(CHARACTERISTIC_UUID_RX, BLECharacteristic::PROPERTY_WRITE | BLECharacteristic::PROPERTY_WRITE_NR);
  pCharacteristicRX->setCallbacks(new MyCallbacks());
  pCharacteristicTX->setCallbacks(new MyCallbacks());
  pCharacteristicTX->addDescriptor(new BLE2902());
  pCharacteristicTX->setNotifyProperty(true);

  // pService = pServer->createService(SERVICE_UUID);
  pCharacteristic1 = pService->createCharacteristic(
      CHARACTERISTIC_UUID,
      BLECharacteristic::PROPERTY_READ |
          BLECharacteristic::PROPERTY_WRITE);
  pCharacteristic1->setValue("Start1");

  pCharacteristic2 = pService->createCharacteristic(
      CHARACTERISTIC2_UUID,
      BLECharacteristic::PROPERTY_READ |
          BLECharacteristic::PROPERTY_WRITE);
  pCharacteristic2->setValue("start2");

  pCharacteristic3 = pService->createCharacteristic(
      CHARACTERISTIC3_UUID,
      BLECharacteristic::PROPERTY_READ |
          BLECharacteristic::PROPERTY_WRITE);
  pCharacteristic3->setValue("start3");

  pCharacteristic4 = pService->createCharacteristic(
      CHARACTERISTIC4_UUID,
      BLECharacteristic::PROPERTY_READ |
          BLECharacteristic::PROPERTY_WRITE);
  pCharacteristic3->setValue("start4");

  pService->start();

  // BLEAdvertising *pAdvertising = pServer->getAdvertising();  // this still is working for backward compatibility
  pAdvertising = BLEDevice::getAdvertising();
  pAdvertising->addServiceUUID(SERVICE_UUID);
  pAdvertising->setScanResponse(true);
  pAdvertising->setMinPreferred(0x06); // functions that help with iPhone connections issue
  pAdvertising->setMinPreferred(0x12);
  BLEDevice::startAdvertising();
  Serial.println("Characteristic defined! Now you can read it in your phone!");

  // pAdvertising->start();
}

void send_BLE(int16_t value, BLECharacteristic *pCharacteristic)
{
  char buffer[80];
  dtostrf(value, 1, 2, buffer);
  pCharacteristic->setValue(buffer);
  pCharacteristic->notify(); // send notification of change
}
```

I used to call in the loop repetitively ... but this has no effect anymore.
```
    pAdvertising->start();
    send_BLE((int16_t)(power1), pCharacteristic1);
    send_BLE((int16_t)(batVoltage_buf[buffer_index]), pCharacteristic2);
    send_BLE((int16_t)(speed_left[buffer_index]), pCharacteristic3);
    send_BLE((int16_t)(charging_state), pCharacteristic4);
```

fbiego, 2021-06-09:
Looks like you have not created `SERVICE_UUID2` on the `pServer`
or if you are adding all characteristic to one service, the changes won't show until you refresh. I use the nRF connect app to refresh the services

goosst, 2021-06-09:
Yeah, no real luck with or without that 2nd one ... . 
I can see both service_UUID's (for a while at least, then it just disappears again untill a reboot of the ESP). 
But I can't connect with the phone to upload a .bin file.

In case you're interested to have a look, I believe to have a minimal example. It sends a value to some characteristics and still implements your OTA method: [main.zip](https://github.com/fbiego/ESP32_BLE_OTA_Arduino/files/6626626/main.zip)

thanks!

fbiego, 2021-06-10:
check this implementation of your code

[main-ota.zip](https://github.com/fbiego/ESP32_BLE_OTA_Arduino/files/6631537/main-ota.zip)

goosst, 2021-06-10:
Great stuff, thanks.
Getting it connected with the app and phone was a bit finicky, but at least it works once I got it connected :+1: 

By the way did you always do this flashing by phone or were there any ways to upload from computer?

fbiego, 2021-06-11:
actually, I'm developing and testing a python script to enable uploads from a computer. I will post it when done

goosst, 2021-06-14:
I've tried again to integrate this example in my own code. 
The best I get is a few % in the upload process in the app, afterwards the ESP just reboots. I'm using the .bin file from your repo as test upload.

```
TX  AA 00 
.TX  F1 00 01 
TX  F1 00 02 
TX  F1 00 03 
E (124461) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (124461) task_wdt:  - IDLE0 (CPU 0)
E (124461) task_wdt: Tasks currently running:
E (124461) task_wdt: CPU 0: btController
E (124461) task_wdt: CPU 1: loopTask
E (124461) task_wdt: Aborting.
abort() was called at PC 0x4015d13c on core 0

ELF file SHA256: 0000000000000000

Backtrace: 0x4008ef34:0x3ffbf6d0 0x4008f1b1:0x3ffbf6f0 0x4015d13c:0x3ffbf710 0x4008be59:0x3ffbf730 0x40161689:0x3ffd3cf0 0x40161c4d:0x3ffd3d10 0x40090216:0x3ffd3d40

```

Is there any debug information that is useful for this? 
I'm about to give up on BLE-OTA.

fbiego, 2021-06-15:
did you modify the `main-ota` code?

goosst, 2021-06-15:
> did you modify the main-ota code?

not the BLE parts: I've compared it a few times, I only changed the order of the function declaration. Latest code is here: 
https://github.com/goosst/hoverboard-firmware-hack-FOC/blob/master/ESP32/src/main.cpp

fbiego, 2021-06-16:
you may want to call your Receive() function in the NORMAL_MODE