fbiego/ESP32_BLE_OTA_Arduino #6: NimBLE

ericlangel, 2021-07-09:
This is not a real issue...more a Feature Request

i ported your ESP32 Code to the ESP32 NimBLE Stack: https://github.com/h2zero/NimBLE-Arduino

first i had an issue with the CRC of the update File, so the Update failed.

Because of this issue i changed the FS to SD Card (SPI Mode) and compared both files.
They are exactly the same and the Update with SD Card is successful.

Now the more Interesting Part:
The Transfer of around 1,2MB Update File was done in 29 Seconds (SD Card) and 43Seconds to FFAT.
And the NimBLE Stack takes less flash Space
 605978 Bytes (46%) with NimBLE Stack
 1031430 Bytes (78%) with ESP32 BLE Stack

Maybe someone knows the reson for failed Checksum when using FFAT?
Trying to start update
Written : 1188512 successfully
E (73242) esp_image: Checksum failed. Calculated 0xd1 read 0x0
Error Occurred. Error #: 9

 
Here is the changed code:
``` 
#include <Update.h>
#include "FS.h"
#include "FFat.h"
#include "SPIFFS.h"
//#include <SD.h>
//#include <BLEDevice.h>
//#include <BLEUtils.h>
//#include <BLEServer.h>
//#include <BLE2902.h>

#include <NimBLEDevice.h>

#define BUILTINLED 2
#define FORMAT_SPIFFS_IF_FAILED true
#define FORMAT_FFAT_IF_FAILED true

//#define USE_SPIFFS  //comment to use FFat

#ifdef USE_SPIFFS
#define FLASH SPIFFS
#define FASTMODE false    //SPIFFS write is slow
#else
#define FLASH FFat
#define FASTMODE true    //FFat is faster
#endif

#define NORMAL_MODE   0   // normal
#define UPDATE_MODE   1   // receiving firmware
#define OTA_MODE      2   // installing firmware

uint8_t updater[16384];
uint8_t updater2[16384];

#define SERVICE_UUID              "fb1e4001-54ae-4a28-9f74-dfccb248601d"
#define CHARACTERISTIC_UUID_RX    "fb1e4002-54ae-4a28-9f74-dfccb248601d"
#define CHARACTERISTIC_UUID_TX    "fb1e4003-54ae-4a28-9f74-dfccb248601d"

static BLECharacteristic* pCharacteristicTX;
static BLECharacteristic* pCharacteristicRX;

static bool deviceConnected = false, sendMode = false;
static bool writeFile = false, request = false;
static int writeLen = 0, writeLen2 = 0;
static bool current = true;
static int parts = 0, next = 0, cur = 0, MTU = 0;
static int MODE = NORMAL_MODE;
[... 291 lines cut ...]
      return;
    }

    size_t updateSize = updateBin.size();

    if (updateSize > 0) {
      Serial.println("Trying to start update");
      performUpdate(updateBin, updateSize);
    }
    else {
      Serial.println("Error, file is empty");
    }

    updateBin.close();

    // when finished remove the binary from spiffs to indicate end of the process
    //Serial.println("Removing update file");
    //fs.remove("/update.bin");

    rebootEspWithReason("Rebooting to complete OTA update");
  }
  else {
    Serial.println("Could not load update.bin from spiffs root");
  }
}

fbiego, 2021-07-10:
Awesome, can you create a PR on a new branch?

ericlangel, 2021-07-12:
yes,

i hope it worked. It's my first PR

fbiego, 2021-07-12:
that was great

closed 2021-07-12 by fbiego