fbiego/ESP32_BLE_OTA_Arduino #8: E (66829) esp_image: invalid segment length 0xffffffff

MarkoHammersen, 2021-07-23:
Dear fbiego
thanks for this great ota example

When I try the "esp32_ble_ota" example I get the error below.
I made sure to have fully erased the flash on my esp32 dev kit c board before uploading the binaries from the Arduino IDE.

Sequence is this:
1. erase flash on esp32 target
2. open "esp32_ble_ota" in Arduino IDE, verify and upload it.
3. open "BLE_OTA_Python-main" in Visual Studio code 
4. python .\ota.py "3C:61:05:3E:C6:2A" "firmware.bin"

Please see the traces below.
Any help to resolve this issue is much appreciated.
thanks
Marko

------------
trace on terminal:
(penv) PS C:\Users\me\Documents\ISE\sensor-devices\esp32\BLE_OTA_Python-main> python .\ota.py "3C:61:05:3E:C6:2A" "firmware.bin"
    ------------------------BLE OTA update---------------------
    Arduino code @ https://github.com/fbiego/ESP32_BLE_OTA_Arduino
#####################################################################
Trying to start OTA update
Reading from:  firmware.bin
Transfer mode: 0
Installing firmware██████████████████████████████████████----| 92.3% Complete
Progress: |██████████████████████████████████████████████████| 100.0% Complete
OTA result:  Written : 160000/160000 [100%]
Error #: 9
Waiting for disconnect... : Device disconnected
-----------Complete--------------
---------------
trace on COM from esp32
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:10944
load:0x40080400,len:6388
entry 0x400806b4
Starting BLE OTA sketch
Characteristic defined! Now you can read it in your phone!
TX  AA 00 
TX  F1 00 01 
TX  F1 00 02 
TX  F1 00 03 
TX  F1 00 04 
TX  F1 00 05 
TX  F1 00 06 
TX  F1 00 07 
TX  F1 00 08 
TX  F1 00 09 
TX  F1 00 0A 
TX  F1 00 0B 
TX  F1 00 0C 
TX  F2 00 0D 
Trying to start update
Written : 160000 successfully
E (66829) esp_image: invalid segment length 0xffffffff
Error Occurred. Error #: 9
TX  0F 57 72 69 74 74 65 6E 20 3A 20 31 36 30 30 30 30 2F 31 36 30 30 30 30 20 5B 31 30 30 25 5D 20 0A 45 72 72 6F 72 20 23 3A 20 39 
Removing update file
Rebooting to complete OTA update
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:10944
load:0x40080400,len:6388
entry 0x400806b4
Starting BLE OTA sketch
Characteristic defined! Now you can read it in your phone!

fbiego, 2021-07-29:
160000? looks like the last part was not written to the file. This usually happens when using SPIFFS.
A delay before installing the update may resolve the issue

Fabchick, 2021-08-11:
I am getting the same error when I am following your tutorial carefully and setting the partition table to "minimal SPIFFS". 
I am working with platform.io plugin on VSCode. There I added `board_build.partitions = min_spiffs.csv` to the platformio.ini file and added this [csv partition table](https://github.com/espressif/arduino-esp32/blob/master/tools/partitions/min_spiffs.csv) to the root in the filesystem.

**But!** everything works when I disregard your advice to set that minimal spiffs partition table. platformio uses the [default partition table](https://github.com/espressif/arduino-esp32/blob/master/tools/partitions/default.csv) which works fine for me.

Sadly, I dont know why it works, because I am new to this and I have to get more knowledge about how those partition stuff works and why I even have to alter it.

ps: fbiego, thanks for the nice tutorial and codes. Really appreciate it because it accelerates my project on OTA over BLE.

fbiego, 2021-08-11:
Hello @Fabchick,
here is a brief description.

`app0` stores the uploaded code via serial and the esp will boot from that by default.
When the OTA file is transferred via BLE it will be stored in `spiffs` temporarily.
The installation process involves transferring the update file from `spiffs` to `app1`, if successful esp will set a flag to boot from `app1` which now contains the new firmware.

So `app0` and `app1` must be greater than your compiled code, your `spiffs` must also be large unless you're using external storage such as an SD card to store the file temporarily

Fabchick, 2021-08-11:
This was an awesome and on-point explanation of it. Thank you! ☺️
Also I have to excuse myself. I have messed up by thinking that you have written [this similar tutorial](https://learn.sparkfun.com/tutorials/esp32-ota-updates-over-ble-from-a-react-web-application/all).

back to topic: 
So If you do not have an SD card, you have to ensure that `app0`, `app1` and `spiffs` is greater than the compiled code.
The "minimum spiffs table" I pointed out earlier has a size of 0x30000 - which is much smaller than `app0` and `app1`.
Could this have been the problem I have got the #9 error?
Also: I just successfully flashed an 725KB binary file via your app to my ESP32. How is this possible when the default partition options give `app0` and `app1` a size of 0x140000? 0x140000 < 725KB binary file, right? Or is the size of the binary file much bigger than the actual size of code transfered?

fbiego, 2021-08-11:
Yeah,  `app0`, `app1` & `spiffs` must be greater.

0x30000 (hex) translates to 196,608 bytes, which was not enough to store the update file

0x140000 is 1,310,720 bytes

Fabchick, 2021-08-12:
Oh, I did not know the hex number is given in bytes. I thought I must divide it by 8 to get the bytes. 😣

You helped me a lot bro! 
Also - I like your projects, keep it up.

commit b0767c3e (2021-12-07): temporary fix #8 #13
closed 2021-12-07 by fbiego