fbiego/ESP32_BLE_OTA_Arduino #9: Device with address is not found

krupis, 2021-09-10:
Hello. There seems to be something missing in your example code. In order to perform firmware update using python script, you must know the device address, but in your program, you havent written a code to print out the device address to the serial monitor which is very inconvenient. I would suggest adding something like :
```
void printDeviceAddress() {
 
  const uint8_t* point = esp_bt_dev_get_address();
 
  for (int i = 0; i < 6; i++) {
 
    char str[3];
 
    sprintf(str, "%02X", (int)point[i]);
    Serial.print(str);
 
    if (i < 5){
      Serial.print(":");
    }
 
  }
}
```

To your code. Additionally, I have managed to get the firmware update using this method only once, I have tried to do it again afterwards and getting this issue: 

```
C:\Users\petrikas.lu\Downloads\BLE_OTA_Python-main\BLE_OTA_Python-main>python ota.py "30:AE:A4:BE:38:32" "firmware.bin"
#####################################################################
    ------------------------BLE OTA update---------------------
    Arduino code @ https://github.com/fbiego/ESP32_BLE_OTA_Arduino
#####################################################################
Trying to start OTA update
-----------Failed--------------
Device with address 30:AE:A4:BE:38:32 could not be found.
```

It seems that it no longer can find the device with such address but my function return this address when I call the function that I have suggested:
```
13:50:57.398 -> Starting BLE OTA sketch
13:50:58.240 -> Characteristic defined! Now you can read it in your phone!
13:50:58.240 -> 30:AE:A4:BE:38:32
```

fbiego, 2021-09-10:
use [`discover.py`](https://github.com/fbiego/BLE_OTA_Python/blob/main/discover.py) to list the available devices

awong1900, 2023-11-06:
If you are use mac, discover.py find list. Then we need use `python ota.py "E202B085-9137-B1E6-DB82-41BAD1694689" "firmware.bin"`
I think people who are not familiar with Bluetooth may not know about this change.

```
yiqiti@TenMac BLE_OTA_Python % python discover.py                              
/Users/yiqiti/Documents/github/BLE_OTA_Python/discover.py:10: DeprecationWarning: There is no current event loop
  loop = asyncio.get_event_loop()
E202B085-9137-B1E6-DB82-41BAD1694689: ESP32 OTA
82E616DE-0EA8-705F-FE41-1409DE4E84F9: Ten
5E36A094-7C9C-7777-7590-398440C9B8E1: None
00C92CBB-A908-2164-F086-54B2BF21FD29: None
3F0B2B1B-E868-36C2-2C34-68DC209F6285: None
8F18A7BD-6D56-5178-BC97-0AB717DAAC16: None
ED44E0CB-F00D-F98C-0749-459D01DBBF61: None
```

closed 2025-04-10 by fbiego