fbiego/ESP32_BLE_OTA_Arduino #14: esp32_ble_ota getting Incomplete while ota

ravisanapati, 2022-01-28:
Hi,
I am  running the esp32_ble_ota code on my esp32 cam board, I have flashed the esp32_ble_ota arduino sketch on my esp32 cam board and then uploaded the bin file using the anrdoid app. In android its is showing file transfered 100%, but in serial monitor it is showing "Incomplete Available space: 603153 File Size: 16596862"
In arduino ide I have selected the following from tools tab Partition Scheme as Default 4MB with spiffs(1.2MBAPP/1.5MB SPIFFS), Board as ESP32 Dev Module.
even I have tried with esp32_nim_ble_ota code too ,but is connecting  and immediately getting disconnecting.
I have attached serial monitor log for esp32_ble_ota, I am not able to find the problem

serial monitor log:

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:10944
load:0x40080400,len:6388
entry 0x400806b4
Starting BLE OTA sketch
Characteristic defined! Now you can read it in your phone!
TX  AA 00 
TX  F1 00 01 
TX  F1 00 02 
TX  F1 00 03 
TX  F1 00 04 
TX  F1 00 05 
TX  F1 00 06 
TX  F1 00 07 
TX  F1 00 08 
TX  F1 00 09 
TX  F1 00 0A 
TX  F1 00 0B 
TX  F1 00 0C 
TX  F1 00 0D 
TX  F1 00 0E 
TX  F1 00 0F 
TX  F1 00 10 
TX  F1 00 11 
TX  F1 00 12 
TX  F1 00 13 
TX  F1 00 14 
TX  F1 00 15 
TX  F1 00 16 
TX  F1 00 17 
TX  F1 00 18 
TX  F1 00 19 
TX  F1 00 1A 
TX  F1 00 1B 
TX  F1 00 1C 
TX  F1 00 1D 
TX  F1 00 1E 
TX  F1 00 1F 
TX  F1 00 20 
TX  F1 00 21 
TX  F1 00 22 
TX  F1 00 23 
TX  F1 00 24 
TX  F1 00 25 
TX  F1 00 26 
TX  F1 00 27 
TX  F1 00 28 
TX  F1 00 29 
TX  F1 00 2A 
TX  F1 00 2B 
TX  F1 00 2C 
TX  F1 00 2D 
TX  F1 00 2E 
TX  F2 00 2F 
Incomplete
Available space: 603153
File Size: 16596862
Incomplete
Incomplete
Incomplete

fbiego, 2022-01-28:
The android app has not been updated, I'll try to do that soon

ravisanapati, 2022-01-29:
thank sfor reply, can you share any working anrdroid app for testing these ota sketches which is working

fbiego, 2022-01-29:
try using FFat, comment this line

https://github.com/fbiego/ESP32_BLE_OTA_Arduino/blob/b0767c3e79a3f774820f66e6d9077916a9fb5f35/esp32_ble_ota/esp32_ble_ota.ino#L39

alternatively, you can check out the python code to update from pc

ravisanapati, 2022-01-31:
python code is working fine thanks for python script, but we have to add the ble part in every new sketch right? and also esp32 should always be in scan mode to check further ota update?
Actually my use case is to do ota from android app I have also tried by using FFat as you suggested, still getting Incomplete could you suggest me how can I solve this issue

ravisanapati, 2022-02-03:
> try using FFat, comment this line
> 
> https://github.com/fbiego/ESP32_BLE_OTA_Arduino/blob/b0767c3e79a3f774820f66e6d9077916a9fb5f35/esp32_ble_ota/esp32_ble_ota.ino#L39
> 
> alternatively, you can check out the python code to update from pc

I have tried using FFat as you suggested still getting Incomplete could you suggest me how can I solve this issue

simogaspa84, 2022-03-09:
Hi i have the same issue of [ravisanapati] but just by using the android app.
With the python code no issues at all.
Can you confirm that commenting line 39 will fix the issue with the android apk?

Thanks a lot

simogaspa84, 2022-03-12:
> thank sfor reply, can you share any working anrdroid app for testing these ota sketches which is working

Did you find a solution to this problem?

closed 2022-04-01 by fbiego

ravisanapati, 2022-04-05:
> > thank sfor reply, can you share any working anrdroid app for testing these ota sketches which is working
> 
> Did you find a solution to this problem?

yes,

#include <Update.h>
#include "FS.h"
#include "FFat.h"
#include "SPIFFS.h"
#include <BLEDevice.h>
#include <BLEUtils.h>
#include <BLEServer.h>
#include <BLE2902.h>

//#define BUILTINLED 33
//#define LED 4
#define FORMAT_SPIFFS_IF_FAILED true
#define FORMAT_FFAT_IF_FAILED true

#define USE_SPIFFS  //comment to use FFat

#ifdef USE_SPIFFS
#define FLASH SPIFFS
#define FASTMODE false    //SPIFFS write is slow  // FC:F5:C4:00:32:52
#else
#define FLASH FFat
#define FASTMODE true    //FFat is faster
#endif

#define NORMAL_MODE   0   // normal
#define UPDATE_MODE   1   // receiving firmware
#define OTA_MODE      2   // installing firmware

uint8_t updater[16384];
uint8_t updater2[16384];

#define SERVICE_UUID              "fb1e4001-54ae-4a28-9f74-dfccb248601d"
#define CHARACTERISTIC_UUID_RX    "fb1e4002-54ae-4a28-9f74-dfccb248601d"
#define CHARACTERISTIC_UUID_TX    "fb1e4003-54ae-4a28-9f74-dfccb248601d"

static BLECharacteristic* pCharacteristicTX;
static BLECharacteristic* pCharacteristicRX;

static bool deviceConnected = false, sendMode = false;
static bool writeFile = false, request = false;
static int writeLen = 0, writeLen2 = 0;
static bool current = true;
static int parts = 0, next = 0, cur = 0, MTU = 0;
static int MODE = NORMAL_MODE;
unsigned long rParts, tParts;

static void rebootEspWithReason(String reason) {
  Serial.println(reason);
  delay(1000);
  ESP.restart();
}

class MyServerCallbacks: public BLEServerCallbacks {
    void onConnect(BLEServer* pServer) {
      Serial.println("Connected");
      deviceConnected = true;

    }
    void onDisconnect(BLEServer* pServer) {
            Serial.println("disConnected");
      deviceConnected = false;
    }
};

[... 302 lines cut ...]
        }
      } 
      Serial.println("in ota mode");  
      
      if (rParts == tParts) {
        //Serial.println("Complete");
        delay(1000);
        //updateFromFS(FLASH);
      } else {
        writeFile = true;
        Serial.println("complete");
        delay(3000);
        updateFromFS(FLASH);
      }  */
      writeFile = true;
      Serial.println("complete");
      delay(3000);
      updateFromFS(FLASH);
      break;

  }

}

try this,its working for me

simogaspa84, 2022-04-05:
Thanks Mr [ravisanapati].. 
Did you try to update the firmware by using the python script after your changes just to see it is still working?
Thanks a lot

ravisanapati, 2022-04-10:
Using python this won't work this works with the android app, the code in the repository..work with python.....there is only few lines change if use the commented lines in OTA_MODE(between /*...*/) Python code will work...if you want update with Android app comment those lines... calling updateFromFS(FLASH);  function at different places is only the difference...for using it python and Android app