fbiego/ESP32_BLE_OTA_Arduino #19: Create simple ArduinoBleOTA library

vovagorodok, 2022-05-24:
Thanks for Your fullstack solution (from esp32 to android app)! I have forked this repository in order to create PlatformIO Arduino library:
https://github.com/vovagorodok/ArduinoBleOTA

At the and will e nice to move it to: https://github.com/arduino-libraries

I'm trying do make it as simple as it possible with reusing of NimBLE-Arduino and ArduinoOTA(support only WIFI) libraries. Proposed interface will be similar to ArduinoOTA:
```
#include <ArduinoBleOTA.h>
void setup() {
  ArduinoBleOTA.begin("ArduinoBleOTA", InternalStorage);  // or SDStorage or SerialFlashStorage
}
```
Or if bluetooth is used not only for OTA:
```
#include <ArduinoBleOTA.h>
void setup() {
    BLEDevice::init("MyBleDevice");
    auto* server = BLEDevice::createServer();
    ArduinoBleOTA.begin(server, InternalStorage);
    // add your BLE services here
    server->startAdvertising();
}
```
Additionally I have plan to add support of Atmel based boards using ArduinoBLE library.

I see that @Vincent-Stragier begin with that. Lets connect our forces and create easy library with simple/lightweight protocol.

Draft protocol proposal will have small number of commands (names and numbers can be another like 0x01, 0xFE):
```
SIZE 1
SIZE_OK 2
SIZE_NOK 3
PACKAGE 4
PACKAGE_ACK 5
INSTALATION_DONE 6
```
Designations for examples:\
`->` - recived from phone\
`<-` - send to phone

Sunny day scenario:
```
-> SIZE <size>
<- SIZE_OK % open ota storage
-> PACKAGE <data>
<- PACKAGE_ACK
...
-> PACKAGE <data>
<- PACKAGE_ACK % full <data> size == <size>, then install
<- INSTALATION_DONE
```

Size to big:
```
-> SIZE <size>
<- SIZE_NOK % do nothing
```

Connection terminated:
```
-> SIZE <size>
<- SIZE_OK
-> PACKAGE <data>
<- PACKAGE_ACK
...
% termination
-> SIZE <size>
<- SIZE_OK % reopen ota storage and write from scratch
...
```
I thing that most/all corner cases are covered or something is missed?
@fbiego can you describe scenarios(corner cases) that  are currently covered with command names (because from code I see some magic numbers 0xFB, 0xFC, 0xFD, FE ..)
What do you think generally about this idea?

Thanks!

Vincent-Stragier, 2022-05-27:
@vovagorodok,

If you take a look at: https://github.com/fbiego/ESP32_BLE_OTA_Arduino/blob/fa90638f2997d7965880c2cd6aca1e9b05c83ba2/esp32_ble_ota_lib_compact/ota_updater.py#L26-L175

Reading the above, you will have a rough idea of how the process works (I'm using a slightly different method compared to the main code).

As you can see, no security mechanism has been implemented in the code. No pin code for the pairing, no checksum at the end of the upload, no version number hard-coded—sometimes, I had a version date and git version hash at compilation time, but not here —, etc. That could also be something to implement in your library.

I do not have much time to help you at the moment, but I like your idea, even if there are still some mechanisms to be thought.

Regards,
Vincent

vovagorodok, 2022-05-27:
Ok, just let me know if you find corner case or bug in my considerations. Will be nice to implement minimal working version and than extend it. For example checksum can be added in future by optional command:
```
-> CHECKSUM <checksum>
<- OK
```
before `INSTALL` command. And developer of mobile app will decide if send it or no. But lets consider it after basic implementation.

I am also interested in @fbiego's point of view

vovagorodok, 2022-09-18:
Hi,\
New library created:
https://github.com/vovagorodok/ArduinoBleOTA \
It has build in checksum integrity protection and software/hardware type/version protection.\
Security/Authorization protection decided to not implement here, because it can be implemented as separate library (for whole bluetooth connection, not for one OTA service).

Library can be simply added to any project like this:\
https://github.com/vovagorodok/ArduinoBleOTA/blob/main/examples/basic/main.ino
Or if there is any additional service:\
https://github.com/vovagorodok/ArduinoBleOTA/blob/main/examples/multiservice/main.ino

Next steps:
1. Add new not ESP32 platforms (will be easy to add `samd` and other using ArduinoBLE library)
2. Increase upload speed. Now I have ~3kB/s. How You achieve ~12kB/s ?
3. Mobile application. Cirrently it updates only by 'updater.py' script. I'm not a mobile developer, some help/integration here will be nice

vovagorodok, 2022-09-19:
According to step 2.\
Added to platformio.ini `board_build.partitions = default_ffat.csv`\
and changed:
https://github.com/vovagorodok/ArduinoBleOTA/blob/061a7ed92c3d64daaddcc865f58a87c455e50823/src/ArduinoBleOTA.cpp#L159 \
to:
```
if (storage == nullptr or not storage->open(firmwareLength, 1)) // second parametr 1
```
it calls with U_SPIFFS:
https://github.com/JAndrassy/ArduinoOTA/blob/ebc6b6299d56813f8565434a10f6d6b978109043/src/InternalStorageESP.cpp#L39 \
And it should call `esp_partition_find_first(...ESP_PARTITION_SUBTYPE_DATA_FAT...)`:
https://github.com/espressif/arduino-esp32/blob/2d6ca351a3b35fe0ce5d427a469e49ed7892c25a/libraries/Update/src/Updater.cpp#L139 \

But update speed is the same (3kB/s).\
From:
https://github.com/espressif/arduino-esp32/blob/master/tools/partitions/default_ffat.csv \
app0 and app1 (ota_0 nad ota_1 SubType) are another partitions, are they fat?\
Are you doing special trick to make it works? Writing to ffat and copy to app0/1 ?

vovagorodok, 2022-09-20:
Step 1. Done. Added new (not `esp32`/`NimBLE-Arduino lib`) platforms like `samd,megaavr,mbed,apollo3,mbed_nano,mbed_portenta,mbed_nicla` based on `ArduinoBLE` library. Plans to add `stm32` and other

Step 2. Done. Works fine with `BLE_OTA_ATTRIBUTE_SIZE=512` and `BLE_OTA_BUFFER_SIZE~=512 * 10`. Can be tuned or buffer can be completely disabled in order to save static memory: https://github.com/vovagorodok/ArduinoBleOTA/blob/main/doc/ADVANCED.md

Step3.  Still open

sdetweil, 2023-06-01:
I was referred  to this URL on BLE performance.. 
https://punchthrough.com/ble-throughput-part-4/
one reported 50k/sec thruput..