fbiego/ESP32_BLE_OTA_Arduino #25: Can't connect to the ESP32

HHavendano, 2023-02-24:
I try to use the code ,and the app. I can select the Firmware file, but i can select de device. 

1. I select the "Chose" button
2. Show me the name ESP32 OTA, and i select it.
3. I wait but nothing happen
4. After 5 seconds, the app closes.

So i can't update the ESP32 for that. Could you help me?

fbiego, 2023-02-24:
I'm looking into this issue

fbiego, 2023-02-25:
the app has been updated

HHavendano, 2023-02-25:
Thanks but still doesn't work.  Select the device and show the esp32 and a few seconds the esp32 disconnect. I'ts about the app or my ESP32?

fbiego, 2023-02-25:
What services and characteristics are broadcasted by ESP32?
You can use `nRF Connect` to view

HHavendano, 2023-02-25:
[image]

HHavendano, 2023-02-25:
I don't know what am i looking for

fbiego, 2023-02-25:
Click on the 3 dots and refresh the services

HHavendano, 2023-02-25:
[image]
Done

fbiego, 2023-02-25:
https://github.com/fbiego/ESP32_BLE_OTA_Arduino/blob/fa90638f2997d7965880c2cd6aca1e9b05c83ba2/esp32_ble_ota/esp32_ble_ota.ino#L56-L58

This service should be listed, after that you can disconnect then use the OTA app

HHavendano, 2023-02-25:
I see, now i can connect it.  now i try to upload the firmware, it downloaded ,but it show me Error# 9. What is it?

fbiego, 2023-02-27:
#26

closed 2023-02-27 by fbiego