fbiego/ESP32_BLE_OTA_Arduino #28: NimBLE OTA and Android app connect and instantly disconnect

PC-maniak, 2023-03-29:
Hi, I'm trying to get "esp32_nim_ble_ota.ino" and the Android app "ESP32 OTA BLE" working,
unfortunately, immediately after connecting to the ESP32, 
"Connected (and one second after) disconnected" appears in the console,
and as a result, the new firmware cannot be uploaded.. the code is not modified in any way .. it's pure code from here and the app comes from the google store 

Please help =( Tom

fbiego, 2023-03-29:
Looks like you had previously connected the esp32 to your phone using different BLE services which might have been cached. You will have to check using nRF Connect app and refresh the services.

PC-maniak, 2023-03-29:
wow that really was it =) sorry, I've only been playing with BT/BLE for a while and I don't know much about it

thank you for your help ;) Tom

PC-maniak, 2023-03-29:
so when I try to OTA update using BLE, I always have an incomplete file =(

Incomplete
Expected: 624384 Received: 615168

on the phone side, it looks like everything is in order .. the only change to the code (for bin for OTA test) is a change to the serial print text

gavin-k-mccormick, 2023-04-10:
@fbiego I am also getting this issue where the ESP32 disconnects almost immediately. I went into nRF connect app and can connected to the ESP32 just fine and then selected 'refresh services'. however, going back to the ESP32 BLE OTA app it still connects for a moment and then disconnects. 

Is there anything else I can try out? this was working last Friday, but now for some reason it started dropping it. I'm using the same Android phone for testing.