fbiego/ESP32_BLE_OTA_Arduino #30: Re-flashing the firmware does not work!

farzadVaziriRaieni, 2023-04-13:
I flash the firmware with Arduino IDE, after that I can use Android App or Python to successfully flash my ESP32-C3 DevKit over BLE. but after that I can't flash the ESP32 anymore. 
I tried both esp32_ble_ota and esp32_nim_ble_ota but it got stuck. do I need to change any setting in Arduino IDE or change the code?
Here is my terminal output, it stuck after this:

Connected
Available space: 1318001
File Size: 573936
@fbiego Can you help me Please!!

farzadVaziriRaieni, 2023-04-17:
I think the problem was the Arduino IDE 2.0.4. I'm using the old IDE 1.8.19 now and it works.
can someone explain why?

simontpellier, 2023-05-17:
Hello farzadVaziriRaieni

I am interested in the soluce !
did you finally solve... without help it seems... this issue ?? (and how !?)
(according to your answer, I intend to try this app... )

simontpellier, 2023-05-17:
really sorry for you
and thanks for the reply
too bad...i give up

simontpellier, 2023-05-17:
... can be a cause?

"... make sure to keep good track of binaries and be doubly sure that a compiled binary has the proper hardware and software version number coded in or you may end up with issues down the road"
(https://learn.sparkfun.com/tutorials/esp32-ota-updates-over-ble-from-a-react-web-application/all)

fbiego, 2023-05-17:
Did you compile the binary file yourself?

farzadVaziriRaieni, 2023-05-17:
Yes, I compiled the code myself.  
I'll look into this again in the near future and report back if I'm successful.

fbiego, 2023-05-17:
which IDE did you use? If Arduino, what version?