fbiego/ESP32_BLE_OTA_Arduino #32: E (106953) esp_image: invalid segment length 0x76727000

my-dudhwala, 2023-08-28:
After upload process, it is showing something related to time. and on serial, I am getting error: E (106953) esp_image: invalid segment length 0x76727000

Full serial output is here,

18:48:55.641 -> Starting BLE OTA sketch
18:48:55.969 -> Characteristic defined! Now you can read it in your phone!
18:49:00.437 -> Connected
18:49:29.869 -> disconnected
18:50:09.226 -> Connected
18:50:22.192 -> TX  AA 00 
18:50:22.192 -> Available space: 113201
18:50:22.192 -> File Size: 625024
18:50:22.511 -> TX  F1 00 01 
18:50:23.059 -> TX  F1 00 02 
18:50:23.725 -> TX  F1 00 03 
18:50:24.331 -> TX  F1 00 04 
18:50:24.842 -> TX  F1 00 05 
18:50:25.449 -> TX  F1 00 06 
18:50:26.010 -> TX  F1 00 07 
18:50:26.333 -> TX  F1 00 08 
18:50:26.659 -> TX  F1 00 09 
18:50:26.984 -> TX  F1 00 0A 
18:50:27.299 -> TX  F1 00 0B 
18:50:27.671 -> TX  F1 00 0C 
18:50:27.988 -> TX  F1 00 0D 
18:50:28.310 -> TX  F1 00 0E 
18:50:28.637 -> TX  F1 00 0F 
18:50:28.960 -> TX  F1 00 10 
18:50:29.330 -> TX  F1 00 11 
18:50:29.611 -> TX  F1 00 12 
18:50:29.939 -> TX  F1 00 13 
18:50:30.265 -> TX  F1 00 14 
18:50:30.592 -> TX  F1 00 15 
18:50:30.873 -> TX  F1 00 16 
18:50:31.196 -> TX  F1 00 17 
18:50:31.522 -> TX  F1 00 18 
18:50:31.849 -> TX  F1 00 19 
18:50:32.174 -> TX  F1 00 1A 
18:50:32.483 -> TX  F1 00 1B 
18:50:32.856 -> TX  F1 00 1C 
18:50:33.181 -> TX  F1 00 1D 
18:50:33.510 -> TX  F1 00 1E 
18:50:33.824 -> TX  F1 00 1F 
18:50:34.198 -> TX  F1 00 20 
18:50:34.524 -> TX  F1 00 21 
18:50:34.795 -> TX  F1 00 22 
18:50:35.124 -> TX  F1 00 23 
18:50:35.499 -> TX  F1 00 24 
18:50:35.826 -> TX  F1 00 25 
18:50:36.141 -> TX  F1 00 26 
18:50:36.316 -> TX  F2 00 27 
18:50:36.399 -> Complete
18:50:41.381 -> Trying to start update
18:50:42.535 -> Written : 100736 successfully
18:50:42.535 -> E (106953) esp_image: invalid segment length 0x76727000
18:50:42.535 -> Error Occurred. Error #: 9
18:50:42.585 -> TX  0F 57 72 69 74 74 65 6E 20 3A 20 31 30 30 37 33 36 2F 31 30 30 37 33 36 20 5B 31 30 30 25 5D 20 0A 45 72 72 6F 72 20 23 3A 20 00 
18:50:47.777 -> Removing update file
18:50:48.010 -> Rebooting to complete OTA update

nguyenlam185, 2024-04-25:
Hello @my-dudhwala ,
I recently got the same problem and still don't know how to proceed further.
Have you fixed this issue? Could you please share me some hints?

Thank you

my-dudhwala, 2024-04-25:
Hi, can you share the issue with me? With the whole discussion is possible?
And what problem I mentioned at that time.

Thanks & Regards